bxparks/AUnit #28: Add ability to do case insensitive string compare

ciband, 2018-05-16:
I am using AUnit with a GTest interface wrapper to allow my unit tests to run on desktop OS as well as Arduino.  I have the following adapter:

#include <AUnit.h>

#define TEST(category, name) test(category##__##name)

#define ASSERT_EQ(e, a) assertEqual(static_cast<decltype(a)>(e), a)
#define ASSERT_NE(e, a) assertNotEqual(static_cast<decltype(a)>(e), a)
#define ASSERT_STREQ(e, a) assertEqual(static_cast<decltype(a)>(e), a)
#define ASSERT_STRNE(e, a) assertNotEqual(static_cast<decltype(a)>(e), a)
#define ASSERT_STRCASEEQ(e, a)  // ??
#define ASSERT_STRCASENE(e, a)  // ??

#define ASSERT_TRUE(x) assertNotEqual(static_cast<unsigned>(x), 0u)
#define ASSERT_FALSE(x) assertEqual(static_cast<unsigned>(x), 0u)

It would be nice to have an easy way to do a case insensitive string compare similar to what GTest does.  If the framework can already do this then a point in the right direction is appreciated.

Thanks for the wonderful framework.

ciband, 2018-05-16:
Dude, you're awesome!  Thank you.  I will pull this ASAP and check it out.

ciband, 2018-05-16:
You need a tip jar. :)  

If you are into the cryptocurrency world, then check out ARK.  
https://ark.io/
https://www.reddit.com/r/ArkEcosystem/

Here is a way to get some free ARK for being a new member of the community.

https://docs.google.com/forms/d/e/1FAIpQLSfkMygnSnSIwRH08H1j1VdZYmp8_zJsYLvRgLYJppgYFQCrzQ/viewform

I can see about getting you a little more for helping out the project.

ciband, 2018-05-16:
Was able to at least verify it compiles for my use case.  Won't be able to test on hardware until tomorrow.  Looks great.

ciband, 2018-05-16:
Yup, we are using it for our C++ wrapper.  Here is our project:

https://github.com/Ark-IoT/Ark-Cpp

Here is a link to the GTest wrapper (such as it is).  It is not complete but you are welcome to copy/use/integrate it into your stuff.

https://github.com/ciband/Ark-Cpp/blob/feature/generate_private_key/test/IoT/gtest/gtest.h

I can also PR it to you if you like but I don't know what kind of organization / interface you would like.

If you do then I'll delete my copy and just use your's.

Thanks again for the extra features.

ciband, 2018-05-18:
This issue can be closed.  Tested the 0.5.3 release and the new functions appear to work well.

The GTest adapter can be tracked in #30.

bxparks, 2018-05-18:
Excellent, thanks for verifying.

closed 2018-05-18 by bxparks