bxparks/AUnit #30: add adapter header file for Google Test

bxparks, 2018-05-16:
In #28, @ciband refers to a Google Test adapter that he wrote:
* https://github.com/ciband/Ark-Cpp/blob/feature/generate_private_key/test/IoT/gtest/gtest.h

It might be useful to other people, so we can host that file in this repo. I propose adding it here:
* `src/aunit/contrib/gtest.h`

People can use it like this:
```
#include <AUnit.h>
#include <aunit/contrib/gtest.h>
```
or
```
#include <AUnitVerbose.h>
#include <aunit/contrib/gtest.h>
```

Questions for @ciband:

* Are you agreeable to using MIT License for your code?
* Why did you use:
    * `#define ASSERT_TRUE(x) assertNotEqual(static_cast<unsigned>(x), 0u)` instead of:
    *  `#define ASSERT_TRUE(x) assertTrue(static_cast<bool>(x))`, or just
    *  `#define ASSERT_TRUE(x) assertTrue(x)` ?

ciband, 2018-05-16:
Totally cool with MIT License

Honestly, because for whatever reason I missed that the assertTrue function/macro existed.  I would be fine with either alternative you suggested.  Much better implementation.

It is worth noting that this gtest.h is not a complete adapter for GTest.  It is only what I've needed so far in our project.  A good example of a hole would be:

std::vector<T> a;
std::vector<T> b;

ASSERT_EQ(a, b);

This works in gtest but will not work with the AUnit / GTest adapter.  Given that this is Arduino, this also may be reasonable and adding support could be grossly out of scope.

bxparks, 2018-05-16:
We probably don't want to support dynamically allocated data structures in the embedded world due to heap fragmentation. Although the ESP32's are getting big and fast enough to they are straddling that grey area.

I just pushed `aunit/contrib/gtest.h` to develop branch. Can you check it out and send me any fixes? Changes from your original version:

* added MIT License with "Copyright (c) 2018 Chris Johnson"
* added doxygen docstring
* changed `ASSERT_TRUE()` and `ASSERT_FALSE()` to use `assertTrue()` and `assertTrue()`, without a `static_cast<>` (hope that works)
* changed `category##_##name` to use only a single underscore, since all identifiers with two underscores (or a leading underscore and Capital) are reserved by the standard

ciband, 2018-05-16:
Sounds great.  I will pull ASAP and let you know.  It will probably be tomorrow.

bxparks, 2018-06-20:
Pushed out with 1.0.0. Closing.

closed 2018-06-20 by bxparks