bxparks/AUnit #35: Comparing float values

gparmeggiani, 2018-07-22:
Hello,

I'm encountering issues when comparing float values, since I get results similar to the following:
`Assertion failed: (44.30) == (44.30)`

My understanding is that AUnit does not provide assert functions specifically designed for floats and doubles (the equivalent of ASSERT_FLOAT_EQ & co of the Google Test framework)

Currently I am using the following workaround:

float epsilon = 0.00001
assertTrue(fabs(44.3 - testValue) < epsilon);

Am I missing something?

bxparks, 2018-07-22:
Hi,

Thanks for using AUnit and posting this question. I should have added some notes about this in the README.md.

You are correct that AUnit does not provide the equivalent of  [ASSERT_FLOAT_EQ and ASSERT_DOUBLE_EQ](https://github.com/google/googletest/blob/master/googletest/docs/advanced.md#floating-point-comparison) from Google Test. The reason is that floating point comparison is subtle and complex. The Google Test ASSERT_FLOAT_EQ() interprets equality to mean 4 ULPs (Units in the Last Place) from each other. The [definition of Units in the Last Place](https://randomascii.wordpress.com/2012/02/25/comparing-floating-point-numbers-2012-edition/) is incredibly complicated.

For embedded environments like Arduino, floating points are not used often so it didn't seem worth spending a lot of time on this.

The `ASSERT_NEAR()` macro from Google Test seems like a much easier to implement and is the same as your workaround. If this functionality is good enough for you, I think I could get that incorporated without too much effort. I would probably name it `assertNear(val1, val2, error)` in AUnit for consistency. I would implement it for floating types, and extend it to integer types if it's straightforward.

gparmeggiani, 2018-07-23:
Yes I think that adding assertNear would be good enough.
I'm now wondering if that could be useful also for integers comparison

bxparks, 2018-07-23:
Please take a look at `assertNear()` and `assertNotNear()` in v1.1 that I just released. They work for integral types, but I suspect that they will be mostly used for floating point types.

PR #36 (2018-07-23): 1.1 - add assertNear() for approximate floating point comparison
* 1.1 (2018-07-23)
    * Add assertNear() and assertNotNear() for floating and integral types.
      Fixes #35.
    * Moved deliberately failing tests into FailingTest.ino and add passing
      tests to AUniter/Jenkins.
closed 2018-07-24 by gparmeggiani