bxparks/AUnit #37: Support assert* in (anonymous) callbacks

squix78, 2018-08-22:
First of all, thank you so much for this great library!!!

I have the following test code:
```
test(parseEvent) {
  String eventText = String("BEGIN:VEVENT\nEND:VEVENT");
  parser.parseEvent(eventText, [](ICalParser::Event event) {
    assertEqual(123, event.startDate);
  });
}
```
Since the assertEqual is called within the callback function I'm getting the following compile error:
```
In file included from .piolibdeps/AUnit_ID2778/src/AUnit.h:53:0,
from src/main.cpp:2:

src/main.cpp: In lambda function:
.piolibdeps/AUnit_ID2778/src/aunit/AssertMacros.h:79:59: error: 'this' was not captured for this lambda function
if (!assertion(__FILE__,__LINE__,(arg1),opName,op,(arg2)))\
^
.piolibdeps/AUnit_ID2778/src/aunit/AssertMacros.h:41:5: note: in expansion of macro 'assertOpInternal'
assertOpInternal(arg1,aunit::internal::compareEqual,"==",arg2)
^
src/main.cpp:56:5: note: in expansion of macro 'assertEqual'
assertEqual(123, 123);
^
*** [.pioenvs/nodemcuv2/src/main.cpp.o] Error 1
```
Is it possible that I can do an assertion in a callback? 
Platform: Espressif/ ESP8266, Platformio

Thank you so much!

bxparks, 2018-10-27:
Hi, Wondering if you had any additional questions. Otherwise, I want to close this for housecleaning.

bxparks, 2018-12-18:
Closing due to no activity for 4 months. Let me know if you have further questions.

closed 2018-12-18 by bxparks