bxparks/AUnit #44: Feature Request:  Support assertIsNull

ciband, 2018-12-03:
Would it be possible to support the "isNull" concept with pointers?  If so, could it play nice with std::nullptr_t such that you can write this:

```
const char* s = nullptr;
assertIsNull(s);
```

Thank you

bxparks, 2018-12-04:
I guess the signature of `assertIsNull()` would be `assertIsNull(const void*)`.

But can you use this instead?
```
assertTrue(s == nullptr);
```
The error message won't be as pretty, but it might be good enough. If the syntax bothers you, maybe we can add a `assertIsNull()` macro to expand to the above `assertTrue()` expression. I want to avoid adding complexity to the core layer if we can avoid it.

ciband, 2018-12-04:
I might have gotten my wires crossed, lol.  I thought GTest has a concept of isnull and thus would be desirable for us to use but now I can't find it.  I think I was thinking of xUnit in C#.  

Nevermind on the assertIsNull.  Sorry for the mix up.

closed 2018-12-04 by ciband

bxparks, 2018-12-04:
I think Google Test uses `ASSERT_EQ(s, nullptr)` which I think will cause problems in AUnit because it wants `assertEqual(s, (const char*) nullptr)`. I have an open issue #34 to look into supporting arbitrary pointers comparisons, because I want to assert on 2 pointers occasionally. I think part of the problem is that AUnit treats `const char*` as special (inherited from ArduinoUnit), to make it easier to compare strings, so I suspect that I'll get overloading problems with `assertEqual(const void*, const void*)`. But I haven't look into this too much.