bxparks/AUnit #45: Feature Request: Support 64 bit integer overloads

ciband, 2018-12-03:
Would it be possible to add overload functions for "long long" and "unsigned long long" for integer comparisions such that this is supported:

uint64_t i = 5;
assertEqual(5ull, i);

Thank you

bxparks, 2018-12-04:
I couldn't support this because [Print](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Print.h) does not support `print(long long)`, so I can't print anything if the assertion fails. However, if you would be happy with a `long long` printed as a 16-character hexidecimal, I think I can add a template specialization in [Assertion.cpp](https://github.com/bxparks/AUnit/blob/develop/src/aunit/Assertion.cpp) to print the `long long` as 2 `long`s. But the avr-gcc will generate a lot of instructions to support 64-bit ints, and you may quickly run out of flash memory, as described in [this forum post](https://forum.arduino.cc/index.php?topic=58697.0).

Just out of curiosity, why do you need 64-bit integers on an 8-bit CPU?

ciband, 2018-12-04:
For our project we use ESP32 and ESP8266, which are 32 bit CPUs.  Some of the protocols we deal with contain uint64_t's.

Are there two issues here:
1) Does the chip support long long and unsigned long long?
2) Does the Arduino framework "print" support 64 bit?

For 1), can we detect that per supported platform and enable the implementation?
For 2), can your implementation convert the 64 bit int to a string for the print instead of having the print function do it?

Thanks for the consideration.

bxparks, 2018-12-18:
Hi Chris,
I had some free time and added support for 64-bit integer types. Do you want to give it a try before I merge it? It's on the `longlong` branch. The following should be supported:

* `assertXxx(long long, long long)`
* `assertXxx(unsigned long long, unsigned long long)`

ciband, 2018-12-22:
Hey Brian,

Sorry for the delay.  Will look at this now.  Thanks for doing this.

ciband, 2018-12-22:
works great on my end.  Thank you.  Tested my use case on esp8266 and esp32.

commit 3d848335 (2018-12-22): Merge branch 'longlong' into develop; fixes #45
closed 2018-12-22 by bxparks