bxparks/AUnit #46: Feature Request: AMock

ciband, 2018-12-03:
Do you have any plans to look into making "AMock", similar to Googletest's GMock?

ciband, 2018-12-04:
For our project at Ark, we will be transitioning to use GMock and would like to keep parity on our IoT builds so a AMock framework with a GMock adapter would be our ideal situation.  When this time comes I will see what I can come up with.  I haven't done my homework on GMock so I don't know how deep the rabbit hole will go.  If I can get something together I will definitely PR it back.

Feel free to close this if you wish or keep it open to nag me, lol. :)

bxparks, 2018-12-04:
I'll leave it open because I am somewhat curious about it (but at a low-level priority). I've not used GMock, but I'm quite familiar with the Java frameworks (JMock, EasyMock, Hamcrest, Mockito, etc) that it's based on.