bxparks/AUnit #50: [BUG] PlatformIO-ESP32 Builds Fail w/AUnit 1.3.

sleepdefic1t, 2019-06-19:
# Issue

AUnit v.1.3 ([#49]) causes builds to fail for PlatformIO and the ESP32.

## examples
**platformio.ini[*failing*]**:
```bash
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:esp32]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = AUnit
upload_speed = 921600
```

**error:**
```bash
.piolibdeps/AUnit_ID2778/unitduino/Arduino.cpp: In function 'long unsigned int millis()':
.piolibdeps/AUnit_ID2778/unitduino/Arduino.cpp:15:17: error: 'CLOCK_MONOTONIC' was not declared in this scope
clock_gettime(CLOCK_MONOTONIC, &spec);
^
.piolibdeps/AUnit_ID2778/unitduino/Arduino.cpp:15:39: error: 'clock_gettime' was not declared in this scope
clock_gettime(CLOCK_MONOTONIC, &spec);
^
*** [.pioenvs/esp32/src/test/.piolibdeps/AUnit_ID2778/unitduino/Arduino.cpp.o] Error 1
```
---

Forcing v.1.2.1 resolved build failures.

**platformio.ini[*successful*]**:
```bash
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:esp32]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = AUnit@1.2.1
upload_speed = 921600
```

## comments

I'm assuming it's something to do with `unitduino`. Not familiar with it though.

## system
- os: **macOS**
- environment: **PlatformIO v.3.6.7**
- board: **ESP32**

bxparks, 2019-06-19:
As a workaround, try removing the entire `AUnit/unitduino` directory, you should be able to make progress. This directory is only meant to be compiled on Linux or MacOS. The Arduino IDE looks only under the `AUnit/src` directory. I guess PlatformIO tries to compile everything under the top level directory? But that doesn't seem right because there are often files under the root directory which are just example snippets of code, which don't necessarily compile.

I haven't played with PlatformIO in a long while. Give me some time to figure out what it's doing.

bxparks, 2019-06-19:
I git pushed a new directory `AUnit/examples/AUnitPlatformIO/` where I tried to reproduce your error. But I am unable to.Everything works fine on my Ubuntu 18.04 machine, with the same PlatformIO version as you (version 3.6.7), the following compiles, uploads and show the test result:
```
$ pio run -e esp32 -t upload --upload-port /dev/ttyUSB1 && \
    pio device monitor -b 115200 -p /dev/ttyUSB1
...
TestRunner started on 1 test(s).
Test exampleTest passed.
TestRunner duration: 0.000 seconds.
TestRunner summary: 1 passed, 0 failed, 0 skipped, 0 timed out, out of 1 test(s).
```

Can you try performing a clean? I think it's `pio run -t clean`. If that doesn't work, I'll try it on my Mac. It'll longer as I don't have PlatformIO installed on my Mac.

sleepdefic1t, 2019-06-20:
Thank you for the response!

After your comments I dug a bit deeper,
and it’s not you.. it’s me 😂

It had to do with our projects layout.
Due to our configuration, the entire `.libdeps` dir was included in the search path during test builds.
This lead to `unitduino/` being included.

It just so happened that none of the prior dependencies had anything that would break builds given our incorrect layout.

This was the perfect motivation to start improving our PIO builds though 👍

Thanks again for your reply,
and keep up the awesome work!!

closed 2019-06-20 by sleepdefic1t

bxparks, 2019-06-20:
Thanks for verifying, good luck with your project!