bxparks/AUnit #56: FeatureRequest: support Arduino Nano Every (ATmega4809)

baumi51, 2020-08-07:
My tests (and example tests) compile correct for Arduino UNO (ATmega328P) but NOT for Arduio Nano Every.

bxparks, 2020-08-14:
Just to let you know that I am not completely ignoring this. I spent some time today scoping the work involved to add support for the megaAVR, and it does *not* seem easy.

The [Arduino megaAVR Core](https://github.com/arduino/ArduinoCore-megaavr/) significantly changed the programming framework in many backwards compatible breaking ways (e.g. https://github.com/arduino/ArduinoCore-API/issues/25). 

I looked at whether it would be possible to use the [MegaCoreX](https://github.com/MCUdude/MegaCoreX) instead. It was said that the "ATmega4809" board should work with the Nano Every. Unfortunately, the MegaCoreX core uses the [same F() hack](https://github.com/arduino/ArduinoCore-megaavr/issues/62) that is used in the Arduino megaAVR Core, which breaks AUnit in non-backwards compatible ways.

I'll poke at this intermittently as time allows, but it is possible that I will decide to make the megaAVR (e.g. the Nano Every) officially **unsupported**.

bxparks, 2021-11-04:
AceTime will not support ArduinoCore-api in the foreseeable future. So it will not support the megaAVR. Closing.

closed 2021-11-04 by bxparks