bxparks/AUnit #60: AUnit freezes with strange arithmetic expression on Nano

brewmanz, 2020-09-26:
AUnit will freeze a Nano (only type tested) several lines before(? strange I know) the following test:
`assertEqual(12345, (int)((long)32767/((long)5 / 9)/2));`

A complete sketch to show freezing is:
```
#include <AUnit.h>

test(T9999_CrashesInArithmeticTest) {
  Print* pPrintDebug = &Serial;
  int xxx;

  if(pPrintDebug) { pPrintDebug->println(); pPrintDebug->flush(); }
  if(pPrintDebug) { pPrintDebug->print(F(" N ")); pPrintDebug->flush(); }
  xxx = 1/0;
  if(pPrintDebug) { pPrintDebug->print(F(" O ")); pPrintDebug->flush(); }
  xxx = (int)((long)12345/((long)5 / 9)/2);
  if(pPrintDebug) { pPrintDebug->print(F(" P ")); pPrintDebug->flush(); }
  xxx = (int)((long)32767/((long)5 / 9)/2);
  if(pPrintDebug) { pPrintDebug->print(F(" Q ")); pPrintDebug->flush(); }
  xxx =              (int)((long)32767/((long)5 / 9)/2);
  if(pPrintDebug) { pPrintDebug->print(F(" R ")); pPrintDebug->flush(); }
#if 1 // Toggle between 0 and 1 and watch the different results
  assertEqual(12345, (int)((long)32767/((long)5 / 9)/2));
#endif
  if(pPrintDebug) { pPrintDebug->print(F(" S ")); pPrintDebug->flush(); }
  assertEqual(1234, 2468);
}

void setup() {
  delay(1000); // wait for stability on some boards to prevent garbage Serial
  Serial.begin(115200); // The baudrate of Serial monitor is set
  while(!Serial); // for the Arduino Leonardo/Micro only
  Serial.println(F(__DATE__ " @ " __TIME__));
  Serial.println(F(__FILE__));

  Print* pPrintDebug = &Serial;
  int xxx;

  if(pPrintDebug) { pPrintDebug->println(); pPrintDebug->flush(); }
  if(pPrintDebug) { pPrintDebug->print(F(" A ")); pPrintDebug->flush(); }
  xxx = 1/2;
  if(pPrintDebug) { pPrintDebug->print(F(" B ")); pPrintDebug->flush(); }
  xxx = 1/0;
  if(pPrintDebug) { pPrintDebug->print(F(" C ")); pPrintDebug->flush(); }
  xxx = (int)((long)12345/((long)5 / 9)/2);
  if(pPrintDebug) { pPrintDebug->print(F(" D ")); pPrintDebug->flush(); }
  xxx = (int)((long)32767/((long)5 / 9)/2);
  if(pPrintDebug) { pPrintDebug->print(F(" E ")); pPrintDebug->flush(); }
}

void loop() {
  aunit::TestRunner::run();
}
```

brewmanz, 2020-09-26:
My output from three runs, without, then with, then without, the failing test line active:
```
21:35:37.904 -> Sep 26 2020 @ 21:35:31
21:35:37.904 -> /home/bryan/git/FastMap/examples/CrashesAUnitMaybe/CrashesAUnitMaybe.ino
21:35:37.904 -> 
21:35:37.904 ->  A  B  C  D  E TestRunner started on 1 test(s).
21:35:37.904 -> 
21:35:37.904 ->  N  O  P  Q  R  S Assertion failed: (1234) == (2468), file /home/bryan/git/FastMap/examples/CrashesAUnitMaybe/CrashesAUnitMaybe.ino, line 21.
21:35:37.937 -> Test T9999_CrashesInArithmeticTest failed.
21:35:37.937 -> TestRunner duration: 0.013 seconds.
21:35:37.937 -> TestRunner summary: 0 passed, 1 failed, 0 skipped, 0 timed out, out of 1 test(s).
21:36:18.729 -> Sep 26 2020 @ 21:36:12
21:36:18.729 -> /home/bryan/git/FastMap/examples/CrashesAUnitMaybe/CrashesAUnitMaybe.ino
21:36:18.729 -> 
21:36:18.729 ->  A  B  C  D  E TestRunner started on 1 test(s).
21:36:18.762 -> 
21:36:18.762 ->  N  O  P Sep 26 2020 @ 21:49:27
21:49:33.075 -> /home/bryan/git/FastMap/examples/CrashesAUnitMaybe/CrashesAUnitMaybe.ino
21:49:33.075 -> 
21:49:33.075 ->  A  B  C  D  E TestRunner started on 1 test(s).
21:49:33.075 -> 
21:49:33.075 ->  N  O  P  Q  R  S Assertion failed: (1234) == (2468), file /home/bryan/git/FastMap/examples/CrashesAUnitMaybe/CrashesAUnitMaybe.ino, line 21.
21:49:33.108 -> Test T9999_CrashesInArithmeticTest failed.
21:49:33.108 -> TestRunner duration: 0.013 seconds.
21:49:33.108 -> TestRunner summary: 0 passed, 1 failed, 0 skipped, 0 timed out, out of 1 test(s).

```

closed 2020-09-26 by brewmanz

bxparks, 2020-09-26:
That's true, division by 0 is UB and the C++ compiler is allowed to do pretty much anything.