bxparks/AUnit #63: Error When Running AUnit: fatal error: AUnit.h: No such file or directory

joesan, 2021-01-01:
I just used this library to write unit tests for my Arduino codebase and I integrated it into GitHub Actions as per the test examples given. When running, I get the following error:

```
MotorControlWithEncoder.ino:6:10: fatal error: AUnit.h: No such file or directory
    6 | #include <AUnit.h>
      |          ^~~~~~~~~
```
Any idea how I could make the GitHub actions know about AUnit being used? In my Makefile, I have the following:

```
APP_NAME := MotorControlWithEncoder
ARDUINO_LIBS := AUnit
include ../../../UnixHostDuino/UnixHostDuino.mk
```

bxparks, 2021-01-02:
Does your Makefile work on your personal computer? If it does, then this is a GitHub Actions configuration issue. But to debug that, I would need to see your github workflow yaml file, which you didn't provide, so I'm going to take blind guesses:

* Remember that GitHub Actions run in its own Docker container, and you have to configure it to download all of its dependencies, including other Arduino libraries. The Makefile itself does not tell GitHub Actions to download those dependencies.
* Take a look at some of my sample workflow files:
    * https://github.com/bxparks/AceTime/blob/develop/.github/workflows
    * https://github.com/bxparks/AUnit/tree/develop/.github/workflows

joesan, 2021-01-03:
I managed to get this fixed by adding a git pull ifor all the dependencies (in my case just the AUnit) n my actions yml file! Thanks for the help!

closed 2021-01-03 by joesan

bxparks, 2021-01-03:
Glad you got it working. Obviously I needed to improve the documentation regarding Continuous Integration options, so I added the following section in the README.md:

https://github.com/bxparks/AUnit#ContinuousIntegration

The additional information that might be relevant for you is that I recommend using the `master` branch instead of the `develop` branch of my libraries, so that you get the stable, released versions, instead of the unstable development versions:
```
git clone --branch master https://github.com/bxparks/UnixHostDuino
git clone --branch master https://github.com/bxparks/AUnit
```