bxparks/AUnit #64: Error When Running AUnit Simple Example

joesan, 2021-01-01:
I created a simple AUnit example as below:

When I ran it via GitHub action, I got the following error:

```
/usr/bin/ld: /tmp/MotorControlWithEncoder.out.m5fOP5.ltrans0.ltrans.o: in function `unixhostduino_main':
<artificial>:(.text+0x4594): undefined reference to `setup'
/usr/bin/ld: <artificial>:(.text+0x45d0): undefined reference to `loop'
collect2: error: ld returned 1 exit status
make[1]: Leaving directory '/home/runner/work/navo-hardware-libs/navo-hardware-libs/test/MotorControlWithEncoder'
make[1]: *** [../../../UnixHostDuino/UnixHostDuino.mk:107: MotorControlWithEncoder.out] Error 1
make: Leaving directory '/home/runner/work/navo-hardware-libs/navo-hardware-libs/test'
make: *** [Makefile:8: tests] Error 2
```

Any ideas what this could mean? Do I need to add a auniter.ini file that contains the configurations? I do not have that yet as I did not find any instructions.

bxparks, 2021-01-02:
Similar to #63, I cannot debug anything without seeing your *.ino and  *yaml files, but I'm going to guess it's the same thing, you need to tell GitHub to install all of its dependencies.

joesan, 2021-01-03:
That was correct! I had to fix this by adding the dependencies to my GitHub actions yaml file.

closed 2021-01-03 by joesan