bxparks/AUnit #69: test macro not working

xavier2910, 2021-02-20:
I am using the `test` macro but I get this error message: `expected an expression`. Is this a problem?
```
#line 2 "Abyz_Fria_Tests.ino"

#include <AUnit.h>

#include "TempSensorDummy.cpp" // testing dummy

test(tempSensor) // error here
{
    TempSensorDummy temp(-1);

    assertEqual(temp.getTemperature(), 20);
    assertEqual(temp.getHumidity(), 20);
    assertEqual(temp.getData(), nullptr);
}

void setup()
{
    Serial.begin(115200);
}

void loop()
{
    aunit::TestRunner::run();
}
```

xavier2910, 2021-02-20:
I am using version 1.5.1

bxparks, 2021-02-20:
What's inside `TempSensorDummy.cpp` file? I'm going to guess that you have a syntax error in there. If you are still having trouble, please verify that the examples under `examples/` directory are working for you. If they are, then it's definitely something in your `TempSensorDummy.cpp`.

xavier2910, 2021-02-23:
Sorry, there was a different problem.

xavier2910, 2021-02-23:
I tried uploading the basic.ino example, but I got this:
```
Arduino: 1.8.12 (Windows 7), Board: "Arduino Uno"

C:\Users\Xavier\AppData\Local\Temp\ccPXerIg.ltrans0.ltrans.o: In function `once':

C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1\examples\basic/basic.ino:15: undefined reference to `aunit::internal::compareNotEqual(int, int)'

C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1\examples\basic/basic.ino:15: undefined reference to `aunit::internal::compareNotEqual(int, int)'

C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1\examples\basic/basic.ino:15: undefined reference to `aunit::Assertion::assertion(char const*, unsigned int, int, char const*, bool (*)(int, int), int)'

C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1\examples\basic/basic.ino:10: undefined reference to `aunit::internal::compareEqual(int, int)'

C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1\examples\basic/basic.ino:10: undefined reference to `aunit::internal::compareEqual(int, int)'

C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1\examples\basic/basic.ino:10: undefined reference to `aunit::Assertion::assertion(char const*, unsigned int, int, char const*, bool (*)(int, int), int)'

C:\Users\Xavier\AppData\Local\Temp\ccPXerIg.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_basic.ino.cpp.o.1819':

<artificial>:(.text.startup+0x5e): undefined reference to `aunit::Test::Test()'

<artificial>:(.text.startup+0x82): undefined reference to `aunit::Test::insert()'

<artificial>:(.text.startup+0x8a): undefined reference to `aunit::Test::Test()'

<artificial>:(.text.startup+0xae): undefined reference to `aunit::Test::insert()'

C:\Users\Xavier\AppData\Local\Temp\ccPXerIg.ltrans0.ltrans.o: In function `run':

C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1/aunit/TestRunner.h:50: undefined reference to `aunit::TestRunner::getRunner()'

C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1/aunit/TestRunner.h:50: undefined reference to `aunit::TestRunner::runTest()'

C:\Users\Xavier\AppData\Local\Temp\ccPXerIg.ltrans0.ltrans.o:(.rodata+0x1a): undefined reference to `aunit::TestOnce::loop()'

C:\Users\Xavier\AppData\Local\Temp\ccPXerIg.ltrans0.ltrans.o:(.rodata+0x26): undefined reference to `aunit::TestOnce::loop()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

xavier2910, 2021-02-23:
with verbose output and all compiler warnings:
```
Arduino: 1.8.12 (Windows 7), Board: "Arduino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Xavier\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10812 -build-path C:\Users\Xavier\AppData\Local\Temp\arduino_build_351687 -warnings=all -build-cache C:\Users\Xavier\AppData\Local\Temp\arduino_cache_58854 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1\examples\basic\basic.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Xavier\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10812 -build-path C:\Users\Xavier\AppData\Local\Temp\arduino_build_351687 -warnings=all -build-cache C:\Users\Xavier\AppData\Local\Temp\arduino_cache_58854 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1\examples\basic\basic.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\Xavier\\AppData\\Local\\Temp\\arduino_build_351687\\sketch\\basic.ino.cpp" -o nul
Alternatives for AUnit.h: [AUnit-1.5.1@1.5.1]
ResolveLibrary(AUnit.h)
  -> candidates: [AUnit-1.5.1@1.5.1]
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Xavier\\Documents\\Arduino\\libraries\\AUnit-1.5.1" "C:\\Users\\Xavier\\AppData\\Local\\Temp\\arduino_build_351687\\sketch\\basic.ino.cpp" -o nul
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Xavier\\Documents\\Arduino\\libraries\\AUnit-1.5.1" "C:\\Users\\Xavier\\AppData\\Local\\Temp\\arduino_build_351687\\sketch\\basic.ino.cpp" -o "C:\\Users\\Xavier\\AppData\\Local\\Temp\\arduino_build_351687\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Xavier\\AppData\\Local\\Temp\\arduino_build_351687\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR " [... cut ...]
C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1/aunit/TestRunner.h:50: undefined reference to `aunit::TestRunner::runTest()'

C:\Users\Xavier\AppData\Local\Temp\cceKOhij.ltrans0.ltrans.o:(.rodata+0x1a): undefined reference to `aunit::TestOnce::loop()'

C:\Users\Xavier\AppData\Local\Temp\cceKOhij.ltrans0.ltrans.o:(.rodata+0x26): undefined reference to `aunit::TestOnce::loop()'

collect2.exe: error: ld returned 1 exit status

Using library AUnit-1.5.1 at version 1.5.1 in folder: C:\Users\Xavier\Documents\Arduino\libraries\AUnit-1.5.1 
exit status 1
Error compiling for board Arduino Uno.
```

bxparks, 2021-02-23:
I'm not sure what's going on with you. I installed Arduino IDE 1.8.13 on my Window 10 machine, and cloned the AUnit v1.5.1 repo into the libraries directory, like you, even renamed it `AUnit-1.5.1`. I cannot reproduce your problem. Do you by any chance have another directory called `AUnit` instead of `AUnit-1.5.1`? Try removing all copies of AUnit and trying again.

xavier2910, 2021-02-26:
I am using VS Code, and VS Code complains about the expression, but when I verify my code it is not a problem

bxparks, 2021-03-03:
Is there anything to be done here? I don't use VS Code on Windows. I cannot reproduce your problem using Arduino IDE.

bxparks, 2021-03-07:
Closing. Please reopen if there's anything to be done.

closed 2021-03-07 by bxparks