bxparks/AUnit #73: Compilation error with gcc 10.x (e.g. RP2040)

timboldt, 2021-05-02:
The RP2040 core uses a newer version of GCC, which encounters the following compilation error:

$ arduino-cli compile --fqbn rp2040:rp2040:adafruitfeather
<homedir>/Arduino/libraries/AUnit/src/aunit/Compare.cpp:123:10: fatal error: WString.h: No such file or directory
  123 | #include <WString.h>
      |          ^~~~~~~~~~~

$ <homedir>/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.1.0-a-81a1771/bin/arm-none-eabi-g++ --version
arm-none-eabi-g++ (GCC) 10.2.0

I'll see if I can figure out why.

bxparks, 2021-05-03:
It's not the compiler. It's a problem with the Arduino API. Which Arduino core are you using? I see:
* https://github.com/arduino/ArduinoCore-mbed
* https://github.com/earlephilhower/arduino-pico

timboldt, 2021-05-03:
I was using https://github.com/earlephilhower/arduino-pico.

It looks like the file was renamed in the Arduino core API:
https://github.com/earlephilhower/ArduinoCore-API/blob/master/api/String.h
vs
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/WString.h

The documentation in the ArduinoCore-API repo says:

```
// including WString.h is deprecated, for all future projects use Arduino.h instead

// This include is added for compatibility, it will be remove on the next
// major release of the API
#include "../String.h"
```

bxparks, 2021-05-03:
It looks like Arduino-Pico (https://github.com/earlephilhower/arduino-pico) is based on the ArduinoCore-API (https://github.com/arduino/ArduinoCore-api). But ArduinoCore-API is backwards incompatible with the previous Arduino API. So Arduino-Pico desperately attempts to provide backwards compatibility, by adding a whole bunch of extra files (e.g. `Print.h`, `IPAddress.h`, `Stream.h`, `Udp.h`), but they are not enough. Even if I manually add a `WString.h`, the new ArduinoCore-API breaks so many things, AUnit does not work with it.

It is unlikely that AUnit or any of my libraries will support the ArduinoCore API in the near future. I don't have the time to track down all the little things that ArduinoCore API breaks. And I also don't have the hardware to test many of these new platforms.

So I have blacklisted the RP2040 and all other platforms using the ArduinoCore-API, so that I can provide a reasonable error message, instead sending the user pages and pages of compiler errors.

bxparks, 2021-05-03:
Pushed out v1.5.5 that contains the blacklist for RP2040.

closed 2021-05-03 by bxparks