bxparks/AUnit #84: Your recommended test for nullptr fails for me. Is this my problem or yours?

brewmanz, 2022-01-07:
When I compile (cf https://github.com/bxparks/AUnit#PointerComparisons)
```
test(nullPointer) {
  const int aa[] = {1, 2};
  assertEqual(aa, nullptr);
}
```
inserted into your `Examples - Aunit - basic`, It fails. (Nano selected)
I believe I'm running your master branch of AUnit f47bd088 from 16 months ago. 
I have a symbolic link in libraries to a local git repo of your AUnit
Is this something bad I've done (I have some local mods to ./arduino15 tables etc), or is the problem 'at your end'?
A work around is replace `nullptr` with `(void*)nullptr`
I get ...
```
/home/bryan/git/Arduino/build/linux/work/arduino-builder -dump-prefs -logger=machine -hardware /home/bryan/git/Arduino/build/linux/work/hardware -hardware /home/bryan/.arduino15/packages -tools /home/bryan/git/Arduino/build/linux/work/tools-builder -tools /home/bryan/git/Arduino/build/linux/work/hardware/tools/avr -tools /home/bryan/.arduino15/packages -built-in-libraries /home/bryan/git/Arduino/build/linux/work/libraries -libraries /home/bryan/git/BrewArduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10815 -build-path /tmp/arduino_build_886393 -warnings=default -build-cache /tmp/arduino_cache_869961 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/home/bryan/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/bryan/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /tmp/arduino_modified_sketch_3246/basic.ino
/home/bryan/git/Arduino/build/linux/work/arduino-builder -compile -logger=machine -hardware /home/bryan/git/Arduino/build/linux/work/hardware -hardware /home/bryan/.arduino15/packages -tools /home/bryan/git/Arduino/build/linux/work/tools-builder -tools /home/bryan/git/Arduino/build/linux/work/hardware/tools/avr -tools /home/bryan/.arduino15/packages -built-in-libraries /home/bryan/git/Arduino/build/linux/work/libraries -libraries /home/bryan/git/BrewArduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10815 -build-path /tmp/arduino_build_886393 -warnings=default -build-cache /tmp/arduino_cache_869961 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/bryan/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/home/bryan/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/bryan/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /tmp/arduino_modified_sketch_3246/basic.ino
Using board 'nano' from platform in folder: /home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3
Using core 'arduino' from platform in folder: /home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3
Detecting libraries used...
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -Werror -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs /tmp/arduino_build_886393/sketch/basic.ino.cpp -o /dev/null
Alternatives for AUnit.h: [AUnit@1.3.3]
ResolveLibrary(AUnit.h)
  -> candidates: [AUnit@1.3.3]
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -Werror -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs -I/home/bryan/git/BrewArduino/libraries/AUnit/src /tmp/arduino_build_886393/sketch/basic.ino.cpp -o /dev/null
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -Werror -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs -I/home/bryan/git/BrewArduino/libraries/AUnit/src /home/bryan/git/BrewArduino/libraries/AUnit/src/aunit/Assertion.cpp -o /dev/null
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -Werror -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs -I/home/bryan/git/BrewArduino/libraries/AUnit/src /home/bryan/git/BrewArduino/libraries/AUnit/src/aunit/Compare.cpp -o /dev/null
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -Werror -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs -I/home/bryan/git/BrewArduino/libraries/AUnit/src /home/bryan/git/BrewArduino/libraries/AUnit/src/aunit/FCString.cpp -o /dev/null
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -Werror -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs -I/home/bryan/git/BrewArduino/libraries/AUnit/src /home/bryan/git/BrewArduino/libraries/AUnit/src/aunit/MetaAssertion.cpp -o /dev/null
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -Werror -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs -I/home/bryan/git/BrewArduino/libraries/AUnit/src /home/bryan/git/BrewArduino/libraries/AUnit/src/aunit/Printer.cpp -o /dev/null
/home/bryan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -Werror -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/bryan/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/bryan/.arduino15/packages/arduino/hard [... cut ...]
In file included from /home/bryan/git/BrewArduino/libraries/AUnit/src/AUnit.h:49:0,
                 from basic.ino:6:
/home/bryan/git/BrewArduino/libraries/AUnit/src/aunit/Assertion.h:68:10: note:   initializing argument 6 of 'bool aunit::Assertion::assertion(const char*, uint16_t, bool, const char*, bool (*)(bool, bool), bool)'
     bool assertion(const char* file, uint16_t line, bool lhs,
          ^~~~~~~~~
Using library AUnit at version 1.3.3 in folder: /home/bryan/git/BrewArduino/libraries/AUnit 
exit status 1
Error compiling for board Arduino Nano.
```

bxparks, 2022-01-07:
The first sentence of https://github.com/bxparks/AUnit#PointerComparisons says that this feature was addedin v1.4. You are using v1.3.3, so it doesn't surprise me that it doesn't work...

brewmanz, 2022-01-07:
Oops. I've been getting a lock failure at V1.3.3 for the past year and never noticed, so fetches via GitEye were doing nothing. Sorry to have wasted your time.

bxparks, 2022-01-10:
Glad you figured it out.

closed 2022-01-10 by bxparks