bxparks/AUnit #85: ASSERT_EQ does not work with refrence

pischky, 2022-01-10:
ASSERT_EQ aunit/contrib/gtest.h does not compile when refrence is used:
````
    uint8_t& ref = a[2];
    ASSERT_EQ( 0x83, ref );
````
Compiler message is:
````
invalid static_cast from type 'int' to type 'uint8_t& {aka unsigned char&}'
````
removing the cast from the macros make make it compile
````
#define ASSERT_EQ(expected, value) assertEqual((expected),(value))
````
May be there is a better (improved) version instead of `static_cast<decltype(a)>(e)`

Complete sketch:
[ArrayTest_with_AUnit.ino.txt](https://github.com/bxparks/AUnit/files/7837463/ArrayTest_with_AUnit.ino.txt)

pischky, 2022-01-10:
May be a solution is to use `remove_reference`:
````
#define ASSERT_EQ(expected, actual)                     \
    assertEqual(                                        \
        static_cast< std::remove_reference<             \
                decltype(actual) >::type >(expected),   \
        actual)
````
But `remove_reference` is not support in current Arduino version (1.8.19) so you need also ArxTypeTraits 0.2.3 by Hideaki Tai.

bxparks, 2022-01-10:
Maybe the solution is to remove the `static_cast<decltype(a)>`? See if the following helps:
https://github.com/bxparks/AUnit/tree/gtest_ref_cast
It fixes your specific problem.

I don't have any code that uses `<gtest.h>`, and I'm not entirely sure I understand all the edge cases of the Google Test framework, so not really sure if this is the best solution.

pischky, 2022-01-10:
Thanks for your suggestion to remove the static_cast. Yes
that fixes my problem. I thought the static_cast has
the intention to select the correct function from
the set of overloaded assertEqual(). I hoped be the original
author of gtest.h did know.

I'm porting some Tests from x86 to arduino. So I'm doing further 
testing with my variant using remove_reference.
So far it seems to work fine. Only disadvantage is that ArxTypeTraits 
is required.

bxparks, 2025-01-20:
I don't know if removing the `static_cast<decltype>` is actually the correct solution, because I don't use `gtest.h`, and I don't have the time/motivation to research this problem any further. So I'm going to close this issue, and delete the `gtest_ref_cast` branch to tidy things up. However, I will copy the code fragment in that branch that seems to solve this problem, for future reference.

```
$ git show e0cdcdc
commit e0cdcdc283495a6b8ae80a460d788d043f90d100 (origin/gtest_ref_cast)
Author: Brian Park <brian@xparks.net>
Date:   Mon Jan 10 07:01:11 2022 -0800

    contrib/gtest.h: Remove static_cast<> which breaks on ASSERT_EQ(0x83, ref) that cast a literal to a reference

diff --git a/src/aunit/contrib/gtest.h b/src/aunit/contrib/gtest.h
index ff26395..b94391f 100644
--- a/src/aunit/contrib/gtest.h
+++ b/src/aunit/contrib/gtest.h
@@ -53,23 +53,21 @@ SOFTWARE.
 
 #define TEST(category, name) test(category##_##name)
 
-#define ASSERT_EQ(e, a) assertEqual(static_cast<decltype(a)>(e), a)
-#define ASSERT_NE(e, a) assertNotEqual(static_cast<decltype(a)>(e), a)
-#define ASSERT_LT(e, a) assertLess(static_cast<decltype(a)>(e), a)
-#define ASSERT_GT(e, a) assertMore(static_cast<decltype(a)>(e), a)
-#define ASSERT_LE(e, a) assertLessOrEqual(static_cast<decltype(a)>(e), a)
-#define ASSERT_GE(e, a) assertMoreOrEqual(static_cast<decltype(a)>(e), a)
+#define ASSERT_EQ(e, a) assertEqual(e, a)
+#define ASSERT_NE(e, a) assertNotEqual(e, a)
+#define ASSERT_LT(e, a) assertLess(e, a)
+#define ASSERT_GT(e, a) assertMore(e, a)
+#define ASSERT_LE(e, a) assertLessOrEqual(e, a)
+#define ASSERT_GE(e, a) assertMoreOrEqual(e, a)
 
-#define ASSERT_STREQ(e, a) assertEqual(static_cast<decltype(a)>(e), a)
-#define ASSERT_STRNE(e, a) assertNotEqual(static_cast<decltype(a)>(e), a)
-#define ASSERT_STRCASEEQ(e, a) \
-    assertStringCaseEqual(static_cast<decltype(a)>(e), a)
-#define ASSERT_STRCASENE(e, a) \
-    assertStringCaseNotEqual(static_cast<decltype(a)>(e), a)
+#define ASSERT_STREQ(e, a) assertEqual(e, a)
+#define ASSERT_STRNE(e, a) assertNotEqual(e, a)
+#define ASSERT_STRCASEEQ(e, a) assertStringCaseEqual(e, a)
+#define ASSERT_STRCASENE(e, a) assertStringCaseNotEqual(e, a)
 
 #define ASSERT_TRUE(x) assertTrue(x)
 #define ASSERT_FALSE(x) assertFalse(x)
 
-#define ASSERT_NEAR(e, a, error) assertNear(static_cast<decltype(a)>(e), a, static_cast<decltype(a)>(error))
+#define ASSERT_NEAR(e, a, error) assertNear(e, a, error)
 
 #endif
```

closed 2025-01-20 by bxparks