bxparks/AUnit #106: Can´t compile my project when using classes with a separate .cpp file

D16174L, 2023-07-28:
I am unable to compile any testing project that tests a class with separate .h and .cpp files. 

I have tried with standard test macros, testF macros, with object definition in several places of the code, and changed around library inclusion order without any effect whatsoever. 

I would suspect of a Linker order problem, but I can't find any makefile to see if this might be. 

By having a single header class definition the issue is solved, yet this is not recommended tipically, as one would want to have separate declarations and definitions. 

Is there any workaround to this. Can this be fixed somehow?

Great project otherwise. Hope to see much more from this.

bxparks, 2023-07-28:
Hi, When asking for technical help, you must provide context about your programming environment if the responder has any chance of answering the question. In your case, you need to provide:

- what hardware are you using? Arduino Nano, ESP8266, ESP32, etc, etc, etc
- which software platform? Arduino AVR Core? Adafruit SAMD Core? Seeeduino SAMD Core? Teensyduino? ESP 8266 Arduino Core? ESP32 Arduino Core? etc etc
- which build tools are you using? Arduino IDE, Arduino CLI, PlatformIO?
- sometimes, you need to provide info about your host operating system: Linux, MacOS
- any additional libraries or dependencies?

Finally, it is often critical to provide a [Minimal Reproducible Example](https://stackoverflow.com/help/minimal-reproducible-example) sample code that reproduces your problem.

Pretend that you are me, and try re-reading what you wrote. Without being able to read your mind, how is it possible for me to answer your question?

D16174L, 2023-07-30:
Hi, thanks for the quick response. 
Hardware: WeMos D1 Mini.
Software Platform: VSCode with Microsoft's Arduino extension using esp8266 arduino Core. 
OS: I am using Windows 10.
No additional Libraries. 

I've been trying to set a basic example to give you a minimal reproducible example, and I've discovered a few things, but AUnit is not the culprit here. However, I will post my findings for anyone interested. I'll try to describe it the best I can. 

I would like my test project structure in the following way:
- headers_under_test
   - test_class.h
   - test_class.cpp
- test.ino

test_class.h:
```
#ifndef TEST_CLASS_H
#define TEST_CLASS_H

class Test_Class 
{
	private: 
		int test_variable;
	public:
		int get_test_variable();
		void test_function();
};

#endif // !TEST_CLASS_H
```
test_class.cpp:
```
#include "test_class.h"

int Test_Class::get_test_variable()
{
	return test_variable;
}

void Test_Class::test_function()
{
	test_variable = 10;
}
```

test.ino:
```
#line 2 "test.ino"

#include <AUnit.h>
using namespace aunit;

#include "hut/test_class.h"

void setup()
{
	// Test Serial Port Setup
	Serial.begin(115200, SERIAL_8N1);
	Printer::setPrinter(&Serial);
	Serial.println("\n");
}

void loop()
{
	TestRunner::run();	
}

test(basic_test_suite, basic_test)
{
	Test_Class test_class;
	test_class.test_function();

	assertEqual(test_class.get_test_variable(), 10);
}
```
If I compile the project with this structure it will fail with the following error:
```
c:/users/joseh/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: d:\Work\test\output\sketch\test.ino.cpp.o:(.text._ZN27basic_test_suite_basic_test4onceEv+0xc): undefined reference to `_ZN10Test_Class13test_functionEv'
c:/users/joseh/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: d:\Work\test\output\sketch\test.ino.cpp.o:(.text._ZN27basic_test_suite_basic_test4onceEv+0x10): undefined reference to `_ZN10Test_Class17get_test_variableEv'
c:/users/joseh/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: d:\Work\test\output\sketch\test.ino.cpp.o: in function `_ZN27basic_test_suite_basic_test4onceEv':
d:\Work\test/test.ino:22: undefined reference to `_ZN10Test_Class13test_functionEv'
c:/users/joseh/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: d:\Work\test/test.ino:22: undefined reference to `_ZN10Test_Class17get_test_variableEv'
collect2.exe: error: ld returned 1 exit status
Using library AUnit at version 1.6.1 in folder: C:\Users\joseh\Documents\Arduino\libraries\AUnit 
exit status 1
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"
[Error] Verifying sketch 'test.ino': Exit with code=1
```

Since I opened this issue I have found two ways to circumvent this:
- by making the class a single header definition which makes the .h unneededly long and harder to read, or
- by changing the project structure pulling the .h and .cpp out of the folder to the root of the project. which makes the test project structure harder to scale. 

Since I've been doing some tests, I have discerned that this issue reproduces whether I use AUnit or not, in any Arduino project. 
If anyone reading this has encountered this issue before and have a better way to fix this, I would be really grateful to know about it. Thanks again.

bxparks, 2023-07-31:
Since this does not seem to be a bug with AUnit, I'm moving this to the Discussions tab.