bxparks/AUnit #114: hi there,

reymalahay, 2023-10-29:
hi there,

what is the correct way to configure the `CPATH` so that all os `aunit`'s header dependencies are correctly referenced? my `CPATH` environment variable is currently configured as follows:

```
export CPATH=/home/reymalahay/Arduino/libraries/AUnit/src
```

but i keep getting the error listed below when running my tests in `EpoxyDuino`.

```
/usr/bin/ld: reymalahay_lab1_code_test.o: warning: relocation against `_ZN5aunit7Printer8sPrinterE' in read-only section `.text._ZN5aunit7Printer10setPrinterEP5Print[_ZN5aunit7Printer10setPrinterEP5Print]'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `test_correct::once()':
reymalahay_lab1_code_test.ino:(.text+0x6b): undefined reference to `aunit::internal::compareEqual(int, int)'
/usr/bin/ld: reymalahay_lab1_code_test.ino:(.text+0x8b): undefined reference to `aunit::Assertion::assertion(char const*, unsigned short, int, char const*, bool (*)(int, int), int)'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `test_incorrect::once()':
reymalahay_lab1_code_test.ino:(.text+0x105): undefined reference to `aunit::internal::compareNotEqual(int, int)'
/usr/bin/ld: reymalahay_lab1_code_test.ino:(.text+0x125): undefined reference to `aunit::Assertion::assertion(char const*, unsigned short, int, char const*, bool (*)(int, int), int)'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `aunit::Printer::getPrinter()':
reymalahay_lab1_code_test.ino:(.text._ZN5aunit7Printer10getPrinterEv[_ZN5aunit7Printer10getPrinterEv]+0xb): undefined reference to `aunit::Printer::sPrinter'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `aunit::Printer::setPrinter(Print*)':
reymalahay_lab1_code_test.ino:(.text._ZN5aunit7Printer10setPrinterEP5Print[_ZN5aunit7Printer10setPrinterEP5Print]+0x13): undefined reference to `aunit::Printer::sPrinter'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `aunit::Test::init(__FlashStringHelper const*)':
reymalahay_lab1_code_test.ino:(.text._ZN5aunit4Test4initEPK19__FlashStringHelper[_ZN5aunit4Test4initEPK19__FlashStringHelper]+0x69): undefined reference to `aunit::Test::insert()'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `aunit::Assertion::Assertion()':
reymalahay_lab1_code_test.ino:(.text._ZN5aunit9AssertionC2Ev[_ZN5aunit9AssertionC5Ev]+0x18): undefined reference to `aunit::Test::Test()'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `aunit::TestOnce::TestOnce()':
reymalahay_lab1_code_test.ino:(.text._ZN5aunit8TestOnceC2Ev[_ZN5aunit8TestOnceC5Ev]+0x1f): undefined reference to `vtable for aunit::TestOnce'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `aunit::TestRunner::run()':
reymalahay_lab1_code_test.ino:(.text._ZN5aunit10TestRunner3runEv[_ZN5aunit10TestRunner3runEv]+0x9): undefined reference to `aunit::TestRunner::getRunner()'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `aunit::TestRunner::runTest()':
reymalahay_lab1_code_test.ino:(.text._ZN5aunit10TestRunner7runTestEv[_ZN5aunit10TestRunner7runTestEv]+0x33): undefined reference to `aunit::TestRunner::printStartRunner() const'
/usr/bin/ld: reymalahay_lab1_code_test.ino:(.text._ZN5aunit10TestRunner7runTestEv[_ZN5aunit10TestRunner7runTestEv]+0x40): undefined reference to `aunit::Test::getRoot()'
/usr/bin/ld: reymalahay_lab1_code_test.ino:(.text._ZN5aunit10TestRunner7runTestEv[_ZN5aunit10TestRunner7runTestEv]+0x79): undefined reference to `aunit::TestRunner::resolveRun() const'
/usr/bin/ld: reymalahay_lab1_code_test.ino:(.text._ZN5aunit10TestRunner7runTestEv[_ZN5aunit10TestRunner7runTestEv]+0xc2): undefined reference to `aunit::Test::getRoot()'
/usr/bin/ld: reymalahay_lab1_code_test.ino:(.text._ZN5aunit10TestRunner7runTestEv[_ZN5aunit10TestRunner7runTestEv]+0x383): undefined reference to `aunit::Test::resolve()'
/usr/bin/ld: reymalahay_lab1_code_test.o: in function `aunit::TestRunner::setupRunner()':
reymalahay_lab1_code_test.ino:(.text._ZN5aunit10TestRunner11setupRunnerEv[_ZN5aunit10TestRunner11setupRunnerEv]+0x42): undefined reference to `aunit::TestRunner::processCommandLine()'
/usr/bin/ld: reymalahay_lab1_code_test.ino:(.text._ZN5aunit10TestRunner11setupRunnerEv[_ZN5aunit10TestRunner11setupRunnerEv]+0x4f): undefined reference to `aunit::TestRunner::countTests()'
/usr/bin/ld: reymalahay_lab1_code_test.ino:(.text._ZN5aunit10TestRunner11setupRunnerEv[_ZN5aunit10TestRunner11setupRunnerEv]+0x5c): undefined reference to `aunit::Test::getRoot()'
/usr/bin/ld: reymalahay_lab1_code_test.o:(.data.rel.ro._ZTV14test_incorrect[_ZTV14test_incorrect]+0x20): undefined reference to `aunit::TestOnce::loop()'
/usr/bin/ld: reymalahay_lab1_code_test.o:(.data.rel.ro._ZTV12test_correct[_ZTV12test_correct]+0x20): undefined reference to `aunit::TestOnce::loop()'
/usr/bin/ld: reymalahay_lab1_code_test.o:(.data.rel.ro._ZTI14test_incorrect[_ZTI14test_incorrect]+0x10): undefined reference to `typeinfo for aunit::TestOnce'
/usr/bin/ld: reymalahay_lab1_code_test.o:(.data.rel.ro._ZTI12test_correct[_ZTI12test_correct]+0x10): undefined reference to `typeinfo for aunit::TestOnce'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make: *** [../EpoxyDuino/EpoxyDuino.mk:210: reymalahay_lab1_code_test.out] Error 1
```

thanks in advance,
rey malahay

_Originally posted by @reymalahay in https://github.com/bxparks/AUnit/discussions/113_

bxparks, 2023-10-29:
Why are you posting duplicate tickets?

closed 2023-10-29 by bxparks

reymalahay, 2023-10-29:
> Why are you posting duplicate tickets?

Sorry, i hit <return> by mistake. i wanted to move this discussion into the `EpoxyDuino` repo, as per your  advice.

apologies,
rey malahay