bxparks/AUnit #116: assertNull and assertNotNull assertions

reymalahay, 2023-10-31:
hi there,

**Is your feature request related to a problem? Please describe.**
this is neither a bug nor a problem, but rather a feature request.

**Describe the solution you'd like**
will it be possible to add explicit `assertNull` and `assertNotNull` assertions? 

**Describe alternatives you've considered**
my current workaround is to use `assertTrue` and `assertFalse`. 

**Additional context**
n/a

please advise.

thanks in advance,
rey malahay

bxparks, 2023-10-31:
Does this help?
https://github.com/bxparks/AUnit#pointer-comparisons

> test(nullPointer) {
>   const int aa[] = {1, 2};
>   assertEqual(aa, nullptr);
> }
> 
> This will print the following:
> 
> AUnitTest.ino:348: Assertion failed: (aa=0x3FFFFF58) == (nullptr=0x0).
> Test nullPointer failed.

I don't remember why 'assertNull()' and 'assertNotNull()' were not included. If I recall, early versions of AUnit had some problems with the incomprehensible C++ casting rules, overloaded function rules, and template type matching rules. Also, GoogleTest does not have assertNull() assertNotNull() (https://google.github.io/googletest/reference/assertions.html), it wants you to use `ASSERT_EQ()`.

I probably would not be opposed to adding an `assertNull()` and `assertNotNull()` as aliases for `assertEqual(x, nullptr)` and `assertNotEqual(x, nullptr)`, if we verify that those work probably in almost all cases.

bxparks, 2023-10-31:
> I probably would not be opposed to adding an `assertNull()` and `assertNotNull()` as aliases for `assertEqual(x, nullptr)` and `assertNotEqual(x, nullptr)`, if we verify that those work ~probably~ properly in almost all cases.

It occurs to me that one reason to create new `assertNull()` and `assertNotNull()` statements, instead of just aliasing them to `assertEqual()`, is that the assertion messages can be shorter and simpler. Otherwise, it would contain the bit about `== (nullptr=0x0)`. So creating new `assertNull()` and `assertNotNull()` would be more work, but definitely better for the end-users.

bxparks, 2023-10-31:
Just for future reference (mostly to myself probably), I think the reason I couldn't add those asserts is due to the ambiguous overloaded functions mentioned in the paragraph right after the one I quoted:

> Comparing a string type (i.e. const char*, or const __FlashStringHelper*) to a nullptr will cause an error due to ambiguous matches on overloaded functions. The solution is to explicitly cast the nullptr to the corresponding string type

So the `assertNull()` and `assertNotNull()` on string types would generate compiler errors. I *think* this can be solved using a `decltype()` statement, which is supposed to be supported in C++11, which I *think* is supported by all Arduino-compatible platforms, but who the heck knows in the Arduino world, so this needs plenty of research and testing.

So, this is more than a trivial amount of work, but hard to say whether it's a small amount or large amount of work.

reymalahay, 2023-10-31:
hi @bxparks ,

thank you for the response.

`assertEqual` works fine. thank you. 

however, `assertNotEqual` times out. this is is sample code that times out when `assertNotEqual` is used:
```
MyObject *myobj;

assertNotEqual(myobj, nullptr);
```

`MyObject` is just an arbitrary class:
```
class MyObject {
}
```

please advise.

thanks in advance,
rey malahay

> Does this help? https://github.com/bxparks/AUnit#pointer-comparisons
> 
> > test(nullPointer) {
> > const int aa[] = {1, 2};
> > assertEqual(aa, nullptr);
> > }
> > This will print the following:
> > AUnitTest.ino:348: Assertion failed: (aa=0x3FFFFF58) == (nullptr=0x0).
> > Test nullPointer failed.
> 
> I don't remember why 'assertNull()' and 'assertNotNull()' were not included. If I recall, early versions of AUnit had some problems with the incomprehensible C++ casting rules, overloaded function rules, and template type matching rules. Also, GoogleTest does not have assertNull() assertNotNull() (https://google.github.io/googletest/reference/assertions.html), it wants you to use `ASSERT_EQ()`.
> 
> I probably would not be opposed to adding an `assertNull()` and `assertNotNull()` as aliases for `assertEqual(x, nullptr)` and `assertNotEqual(x, nullptr)`, if we verify that those work ~probably~ properly in almost all cases.

bxparks, 2023-10-31:
I cannot offer any help if the only thing you give me is "it times out". You need to provide: sample code, the environment, the compiler, the commands used... Just any little thing that explains how to reproduce your problem.

reymalahay, 2023-10-31:
hi @bxparks ,

as requested:

my test:
```
#line 2 "lab1_code_test.ino"

#include "../src/MyObject.cpp"
#include <AUnit.h>

testing(myobject_valid) {

    MyObject *myobj;

    assertNotEqual(myobj, nullptr);
}

void setup() {

}

void loop() {
  aunit::TestRunner::run();
}

```

MyObject.cpp
```
class MyObject {
};

```

environment: Linux

compiler: /usr/bin/g++

commands used:
```
make clean
make
make run
```

thanks in advance,
rey malahay

> I cannot offer any help if the only thing you give me is "it times out". You need to provide: sample code, the environment, the compiler, the commands used... Just any little thing that explains how to reproduce your problem.

bxparks, 2023-11-02:
> testing(myobject_valid) {

Replace `testing()` with `test()`. The `testing()` macro runs forever.

closed 2023-11-02 by bxparks