bxparks/AUnit #125: Suggestion: Add assertStringContains (etc) testing options.

brewmanz, 2025-01-17:
I have defined a macro
```
#define assertStringContains(arg1,arg2) \
    assertOpVerboseInternal(arg1,aunit::internal::compareSubstring,\
    "(contains)",arg2)
```
which nicely does what I want, as the `internal::compareSubstring` already exists.
e.g.
```
 assertStringContains(myLongErrMsg, "Average Speed");
 assertStringContains(myLongErrMsg, "divide by zero");
```

A) Maybe I've overlooked an existing way to do it (easily). If so, sorry that I didn't look harder.
B) A non-verbose macro would also be needed.
C) A bonus might be `assertStringNotContains`.
D) `assertStringCase[Not]Contains`, `assertString[Case][Not]StartsWith` and `assertString[Case][Not]EndsWith` might be handy too.

bxparks, 2025-01-20:
I guess you are pointing out that the `internal::compareSubstring()` functions in `Compare.h` are not exposed through a corresponding macros in `AssertMacros.h`.

I think what happened was that I needed the `internal::compareString()` function so that I could implement the `assertXXX()` for string types .

Then I needed the `internal::compareStringN()` to implement the prefix-string matching feature in `TestRunner::include()` and `exclude()`. 

Then I implemented `internal::compareSubstring()` functions just for completeness. Maybe. I don't remember. It looks like I didn't need them for anything else, so I probably forgot to expose them through the `assertXXX()` macros.

Maybe they are worth exposing through `assertXXX()`. Maybe not. I'm not sure. My usual workaround for people who need more complex comparison operators is to use `assertTrue()` and `assertFalse()`.

The concern that I have with some of the other assertions that you mention (e.g. assertStringCase[Not]Contains, assertString[Case][Not]StartsWith and assertString[Case][Not]EndsWith) is the combinatorial explosion of such comparision operators. Where does it end? I think the way Google Test and other unit test frameworks solve this problem is to off load the comparison operators to *another* framework or library. That's more complexity than I want to deal with.

The current situation is that I am not doing much Arduino programming these days. I've moved to other projects which will probably occupy my attention for the next several years. I will probably come back to Arduino stuff in the future, but right now, I don't want really have the time or motivation.

If you are interested in implementing this in AUnit, then maybe you can volunteer to become a co-maintainer of AUnit? Be warned, it is not just the coding. There is a fair bit of work in testing, the validation against all the various Arduino-compatible boards, and the release engineering.

brewmanz, 2025-01-20:
Whilst I'm occasionally active with Arduino Dev stuff, my availability is patchy too.

bxparks, 2025-01-20:
Then I'm closing this as "Not Planned".

closed 2025-01-20 by bxparks